ComNets-Bremen/MoleNet #19: Low power / wakeup

Wolfestin123, 2024-02-22:


jdede, 2024-02-22:
Please add some content: Is the low power working or failing?

Wolfestin123, 2024-02-29:
The power consumption measurement of MoleNet V6 is attached in the pdf below. 
[MoleNet - V6_power_consumption.pdf](https://github.com/ComNets-Bremen/WUSN/files/14448364/MoleNet.-.V6_power_consumption.pdf)
The code used :
`
from machine import deepsleep
from machine import lightsleep
from time import sleep

print('Im awake. Going to sleep in 10 seconds')
sleep(10)
print('Going to 10 sec light sleep now')
lightsleep(10000)
sleep(10)
print('Going to 10 sec deep sleep now')
deepsleep(10000)

`

jdede, 2024-02-29:
2 mA with LoRa and SD powered up is too high. Please set the LoRa-module into deep sleep mode and check the current drawn by the pull-up resistors.

Wolfestin123, 2024-04-03:
I have put the LoRa module to sleep and measured the current consumption. The results indicate a reduction in current consumption of the LoRa chip (RA-01H) by **2mA** during sleep. Please find the attached code and results for reference.
[MoleNet - V6_power_consumption_with_LORA_sleep.pdf](https://github.com/ComNets-Bremen/WUSN/files/14852768/MoleNet.-.V6_power_consumption_with_LORA_sleep.pdf)

`import LoRaSender
import LoRaReceiver

from config import *
from machine import Pin, SPI, SoftSPI
from sx127x import SX127x
from machine import lightsleep
from time import sleep
from machine import deepsleep

device_spi = SoftSPI(
    baudrate = 200000, 
    polarity = 0, phase = 0, bits = 8, firstbit = SPI.MSB,
    sck = Pin(device_config['sck'], Pin.OUT, Pin.PULL_DOWN),
    mosi = Pin(device_config['mosi'], Pin.OUT, Pin.PULL_UP),
    miso = Pin(device_config['miso'], Pin.IN, Pin.PULL_UP))

Pin(device_config['reset'], Pin.OUT, value=1)
lora = SX127x(device_spi, pins=device_config, parameters=lora_parameters)

print("going to light sleep")

lightsleep(10000) # 10 seconds light sleep
print("lora on")
sleep(20)

lora.sleep()  # Lora is on sleep mode
print("lora off")

lightsleep(10000) # 10 seconds light sleep

deepsleep(20000) # 20 seconds deep sleep

`

jdede, 2025-05-12:
Correct me if I'm wrong: The reason for this issue was the wrong voltage regulator. With the newer version, this should be fine. @Wolfestin123 @FarukKollar @B00mb0x can you confirm?